turgu1/EPub-InkPlate #1: Unable to get Solaris epub to work

aereaux, 2021-03-12:
Thank you for writing this software, it works great for most of the ebooks I have been trying.  Unfortunately, I am having trouble getting the epub [here](https://b-ok.cc/dl/6109258/eb9a52) to work.  It just gets stuck on the "Loading a book" screen (and it works fine in other ebook readers on my desktop and phone).  Is there any way to get any more debug output in order to figure out what is going wrong?

Thanks in advance, let me know if any more information would help

turgu1, 2021-03-12:
Hello,
Can you indicate which version of the application you are using?
Thanks.

turgu1, 2021-03-12:
I've tried to load your book with the current version of the application (v1.2.0) and I'm able to load it and read it on my own version of the InkPlate-6.

aereaux, 2021-03-12:
Hello, I'm using version v1.2.0 of the application.  Thanks for working on reproducing it, I'll try reinstalling and see if it still occurs.

turgu1, 2021-03-12:
If it still not working, please send me you current display parameters as they appears in the "Main Parameters" and "Default e-books parameters" forms.

turgu1, 2021-03-12:
Another possibility would be that your e-book was not loaded properly on the SDCard. If it's not working, you can try to delete it and reload it.

aereaux, 2021-03-13:
I have tried reformatting and copying the SDCard multiple times and it doesn't seem to work.  I used to have micropython installed on this inkplate, is there something I should try to do to reset it?  The settings should be the default after installation, but if that doesn't help, I can send them as they appear.  On a first load of the ebook, I get errors like this, not sure if they're relevant:
```
E (18320) PageLocs: Unable to open pages location file.

E (20445) CSS: display not decoded: 'display' at offset: 70

E (20445) CSS: display not decoded: 'display' at offset: 18

E (20445) CSS: display not decoded: 'display' at offset: 19

E (20455) CSS: text-align not decoded: 'inherit' at offset: 138

E (20465) CSS: display not decoded: 'display' at offset: 18

E (20575) TTF: Charcode not found in face: 322, font_index: 1

E (42655) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:

E (42655) task_wdt:  - IDLE0 (CPU 0)

E (42655) task_wdt: Tasks currently running:

E (42655) task_wdt: CPU 0: retrieverTask

E (42655) task_wdt: CPU 1: IDLE1

E (42655) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x401095C3:0x3FFB0D80 0x400862ED:0x3FFB0DA0 0x4000BFED:0x3FFF2C30 0x4009025F:0x3FFF2C40 0x40093C11:0x3FFF2C60 0x400865B1:0x3FFF2C80 0x400865F4:0x3FFF2CA0 0x4009708D:0x3FFF2CC0 0x401B9C35:0x3FFF2CE0 0x400ED9CD:0x3FFF2D00 0x400EED5D:0x3FFF2D20 0x400DF834:0x3FFF2D70 0x400DF900:0x3FFF2E80 0x400DF900:0x3FFF2F90 0x400DF900:0x3FFF30A0 0x400DFBBD:0x3FFF31B0 0x400DFC64:0x3FFF3250 0x401CD9BD:0x3FFF3290 0x4012FE18:0x3FFF32B0 0x4008FFB5:0x3FFF32D0

E (42655) task_wdt: Print CPU 1 backtrace

Backtrace:0x40083046:0x3FFB1380 0x400862ED:0x3FFB13A0 0x4000BFED:0x3FFC10A0 0x4009025F:0x3FFC10B0 0x40109887:0x3FFC10D0 0x40109893:0x3FFC1100 0x40109961:0x3FFC1120 0x40090C99:0x3FFC1140 0x4008FFB5:0x3FFC1160
```

turgu1, 2021-03-13:
Hello Aidan,

The error msgs are OK. They are related to some aspects of the document that the parser is not taking into account and some issue on the watchdog management that I'm aware of. I get the same msgs on my own device, but I still can see the pages of your book without any problem.

One think that can help me: can you zip the content of the SD Card and send it to me through Dropbox or other mean? I would then be able at least to check if there is something weird there.

Guy

aereaux, 2021-03-13:
[sdcard.zip](https://github.com/turgu1/EPub-InkPlate/files/6135281/sdcard.zip)
Here are the contents of the sdcard

turgu1, 2021-03-13:
I may have found the problem. It's a bug that I will investigate. In the "Default e-books parameters" form, the "Show Images in E-Books" is set to NO. If you change it to YES, you will be OK to see the content of the book.

aereaux, 2021-03-13:
Thanks a lot!  That seems to work.

turgu1, 2021-03-13:
You are welcome! And thanks to have reported the problem.

aereaux, 2021-03-13:
One other quick question: you can't use the touch buttons with the 3d printed case provided with the inkplate, right?  What sort of case do you use or suggest?

turgu1, 2021-03-13:
The provided case does not supply access to the touch buttons. I've seen in the forum some people that managed to get access to them doing some small modification to the case with some screws or little pieces of  a metallic foil: [here](https://forum.e-radionica.com/en/viewtopic.php?f=23&t=266).

Another solution is to replace the touch buttons with mechanical buttons that would be located on the surface of the supplied case.

I've designed a solution: an extended case with a circuit board that allow to add 6 mechanical buttons. You can find all the information [here](https://forum.e-radionica.com/en/viewtopic.php?f=24&t=325). That requires that you have access to a 3D printer and ready to spend some $$ to acquire the circuit board and the buttons...

aereaux, 2021-03-13:
Thanks for the suggestions.  I quickly tried the aluminum foil method and wasn't able to get it to work, but I'll keep experimenting.

turgu1, 2021-03-16:
The issue has been corrected in v1.2.1

closed 2021-03-16 by turgu1