turgu1/EPub-InkPlate #2: Small issue when trying to idf.py build

martinberlin, 2021-03-14:
Very nice project!
I'm developing an [epaper component for the IDF](https://github.com/martinberlin/cale-idf) and I though would be very nice to make an ePub reader that support multiple epapers. But I see you made a great work here so I guess will start using yours. 

Tried to compile, my idf version is this one:
ESP-IDF v4.2-dev

```
/home/martin/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x24): undefined reference to `app_main'
/home/martin/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/esp32/libesp32.a(cpu_start.c.obj): in function `main_task':
/home/martin/esp/esp-idf/components/esp32/cpu_start.c:567: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
```

Or I'm building this wrong and I have to build it using Platformio? Thanks in advance

turgu1, 2021-03-14:
Hello Martin,
If you are compiling an application, ESP-IDF requires that an
app_main() function be declared as a C function to be called by the
SDK. This is the main entry point to your application. As it must be a
C function if you are using C++, you have to encapsulate it as follows:
  extern "C" {
    void     app_main(void)    {
    }
  } // extern "C"

Usually, the ESP-IDF is compiled at the same time as an application
that requires it. You cannot create, say, a pre-compiled library with
it.
Hope this help
Cheers!
Guy

turgu1, 2021-03-14:
I may now better understand your question :-)
The EPub-Inkplate was developped using PlaformIO. I don't expect that
you could get it built directly with ESP-IDF unless you modify the
folder structure or add CMake elements in the built description files.

turgu1, 2021-03-14:
Some more information:
If you look inside the platformio.ini file, you will see that there is
a need for many specifics aspects to be defined to get the EPub-Reader
compiled. The message you get is related to the fact that some #defined
values are not properly setup. As the application as been designed to
support building both linux and esp32 based executables, a proper
#define value must be set through the c++ -D option to select the
targetted device.
For the standard InkPlate6 device, look in platformio.ini config file
for the following header:
[env:inkplate_6_release]

martinberlin, 2021-03-14:
Thanks a lot for the guidance. The fact is that I use Platformio also but only as an IDE and then I build and compile my Firmware's directly in command-line.

Now I'm trying to build it with platformio, updated the platformio default_envs

```
[platformio]
default_envs = inkplate_6_release
```

Getting:
In file included from include/models/fonts.hpp:8,
                 from include/models/css.hpp:10,
                 from include/models/epub.hpp:11,
                 from include/controllers/books_dir_controller.hpp:9,
                 from src/controllers/app_controller.cpp:8:
include/models/ttf2.hpp:11:10: fatal error: ft2build.h: No such file or directory

But file is there, seems just not where Platformio is trying to search for it:
EPub-InkPlate$ find . -name ft2build.h
./freetype-2.10.4/devel/ft2build.h
./freetype-2.10.4/include/ft2build.h
./lib_freetype/include/freetype2/ft2build.h

Sorry I know that I'm a bit lost here. Maybe there is some build command that I forgot to run, or an install section that I forgot to read?

turgu1, 2021-03-14:
Did you grab the complete platformio.ini file from the github
repository? The inkplate_6_release env retuires access to other
definitions under [inkplate_common] and [common].
Guy

martinberlin, 2021-03-14:
Sure, I cloned this repository completely, so I have the same stand as you in master. Thanks a lot will research a bit more

turgu1, 2021-03-14:
I'm using that configuration without any change on two different
computer: One running Ubuntu 20.04, another running Linux Mint V20. 
I will do some test on a VmWare virtual Ubuntu in case there is
something I've missed....
Guy

turgu1, 2021-03-14:
I've just built successfully the EPub-InkPlate reader using a vanilla
Linux Ubuntu 20.04 system in a virtual VMPlayer device. I've done the
following:
1. installed git using: sudo apt install git2. cloned the github
project using: git clone https://github.com/turgu1/EPub-InkPlate.git3.
Installed VSCode using the Ubuntu Software management tool4. installed
the python3 distutils tool using: sudo apt install python3-distutils5.
started vscod and installed platformio6. opened the EPub-Inkplate
folder in vscode7. launched the build. 
You will find attached the end of the output from the building process
(I've lost the start of it... too long for the VSCode display).
Hope that help.
Guy

Archiving .pio/build/inkplate_6_release/esp-idf/esp32/libesp32.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp32/libesp32.a
Compiling .pio/build/inkplate_6_release/esp_common/src/dbg_stubs.o
Compiling .pio/build/inkplate_6_release/esp_common/src/esp_err_to_name.o
Compiling .pio/build/inkplate_6_release/esp_common/src/freertos_hooks.o
Compiling .pio/build/inkplate_6_release/esp_common/src/mac_addr.o
Compiling .pio/build/inkplate_6_release/esp_common/src/pm_locks.o
Compiling .pio/build/inkplate_6_release/esp_common/src/stack_check.o
Compiling .pio/build/inkplate_6_release/esp_common/src/task_wdt.o
Compiling .pio/build/inkplate_6_release/esp_common/src/int_wdt.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth.o
Archiving .pio/build/inkplate_6_release/esp-idf/esp_common/libesp_common.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp_common/libesp_common.a
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_phy.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_netif_glue.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_mac_esp32.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_phy_dp83848.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_phy_ip101.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_phy_lan8720.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_phy_rtl8201.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_mac_dm9051.o
Compiling .pio/build/inkplate_6_release/esp_eth/src/esp_eth_phy_dm9051.o
Compiling .pio/build/inkplate_6_release/esp_event/default_event_loop.o
Archiving .pio/build/inkplate_6_release/esp-idf/esp_eth/libesp_eth.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp_eth/libesp_eth.a
Compiling .pio/build/inkplate_6_release/esp_event/esp_event.o
Compiling .pio/build/inkplate_6_release/esp_event/esp_event_private.o
Compiling .pio/build/inkplate_6_release/esp_event/event_loop_legacy.o
Compiling .pio/build/inkplate_6_release/esp_event/event_send.o
Compiling .pio/build/inkplate_6_release/esp_gdbstub/src/gdbstub.o
Archiving .pio/build/inkplate_6_release/esp-idf/esp_event/libesp_event.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp_event/libesp_event.a
Compiling .pio/build/inkplate_6_release/esp_gdbstub/src/packet.o
Compiling .pio/build/inkplate_6_release/esp_gdbstub/esp32/gdbstub_esp32.o
Compiling .pio/build/inkplate_6_release/esp_gdbstub/xtensa/gdbstub_xtensa.o
Compiling .pio/build/inkplate_6_release/esp_hid/src/esp_hidd.o
Archiving .pio/build/inkplate_6_release/esp-idf/esp_gdbstub/libesp_gdbstub.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp_gdbstub/libesp_gdbstub.a
Compiling .pio/build/inkplate_6_release/esp_hid/src/esp_hidh.o
Compiling .pio/build/inkplate_6_release/esp_hid/src/esp_hid_common.o
Compiling .pio/build/inkplate_6_release/esp_http_client/esp_http_client.o
Archiving .pio/build/inkplate_6_release/esp-idf/esp_hid/libesp_hid.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp_hid/libesp_hid.a
Compiling .pio/build/inkplate_6_release/esp_http_client/lib/http_auth.o
Compiling .pio/build/inkplate_6_release/esp_http_client/lib/http_header.o
Compiling .pio/build/inkplate_6_release/esp_http_client/lib/http_utils.o
Compiling .pio/build/inkplate_6_release/esp_http_server/src/httpd_main.o
Archiving .pio/build/inkplate_6_release/esp-idf/esp_http_client/libesp_http_client.a
Indexing .pio/build/inkplate_6_release/esp-idf/esp_http_client/libesp_http_client.a
Compiling .pio/build/inkplate_6_release/esp_http_server/src/httpd_parse.o
Compiling .pio/build/inkplate_6_release/esp_http_server/src/httpd_sess.o
Compiling .pio/build/inkplate_6_release/esp_http_server/src/httpd_txrx.o
Compiling .pio/build/inkplate_6_release/esp_http_server/src/httpd_uri.o
Compiling .pio/build/inkplate_6_release/esp_http_server/src/httpd_ws.o
Compiling .pio/build/inkplate_6_release/esp_http_server/src/util/ctrl_sock.o
Compiling .pio/build/inkplate_6_release/esp_https_ota/src/esp_https_ota.o
[... 753 lines cut ...]
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/rsa.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/rsa_internal.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/sha1.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/sha256.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/sha512.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/threading.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/timing.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/version.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/version_features.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/xtea.o
Compiling .pio/build/inkplate_6_release/mbedtls/port/esp_hardware.o
[... cut ...]
Compiling .pio/build/inkplate_6_release/mbedtls/port/esp_bignum.o
Compiling .pio/build/inkplate_6_release/mbedtls/port/esp32/bignum.o
Compiling .pio/build/inkplate_6_release/mbedtls/port/esp32/esp_sha1.o
Compiling .pio/build/inkplate_6_release/mbedtls/port/esp32/esp_sha256.o
Compiling .pio/build/inkplate_6_release/mbedtls/port/esp32/esp_sha512.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/debug.o
Compiling .pio/build/inkplate_6_release/mbedtls/mbedtls/library/ssl_cache.o
Compiling .pio/b

turgu1, 2021-03-14:
Another point: be sure using the platformio build task through the
build icon at the bottom of the VSCode window...
Guy

turgu1, 2021-03-14:
I've just seen your update comment where you tried to compile the freetype library (I was reading your comments through my email and the last part of your msg was not sent to me...). As mentioned in the readme, this is not required. It's already done and everything is at the right place in the folder structure.
I suggest that you retrieve a fresh copy of the GitHub project and simply start a single platformio build command.

martinberlin, 2021-03-14:
Ok that’s it I will start from zero again tomorrow. Thanks a lot for the effort.
Maybe this is because I use esp-IDF separately of PlatformIO so maybe there is some conflict. Maybe doing this with a clean Ubuntu is a great idea.

turgu1, 2021-03-14:
There is no problem of having a other ESP-IDF SDK installed as long as
your system path doesn't point at it I gues. I have such a SDK on my
laptop and it is not used at all.
Platformio take a copy of the sdk from the net and push it in a
subfolder in the .pio folder under the main project folder and compile
it there.

martinberlin, 2021-03-14:
Pulled it again from 0 repeating your steps (Except OS is the same Ubuntu 20)
This is what I get:

```
Processing inkplate_6_release (platform: espressif32; board: esp-wrover-kit; framework: espidf)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (1.12.4) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40001.200521 (4.0.1) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.190618 (2.28.51) 
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP-IDF-Inkplate> 0.9.4+sha.ef3a5f8
|-- <EPub_InkPlate>
|   |-- <ESP-IDF-Inkplate> 0.9.4+sha.ef3a5f8
|-- <externals>
|-- <tools>
Building in release mode
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/app_controller.cpp.o
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/book_controller.cpp.o
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/books_dir_controller.cpp.o
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/common_actions.cpp.o
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/event_mgr.cpp.o
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/option_controller.cpp.o
Compiling .pio/build/inkplate_6_release/esp-idf/src/controllers/param_controller.cpp.o
In file included from include/models/fonts.hpp:8,
                 from include/models/css.hpp:10,
                 from include/models/epub.hpp:11,
                 from include/controllers/books_dir_controller.hpp:9,
                 from src/controllers/app_controller.cpp:8:
include/models/ttf2.hpp:11:10: fatal error: ft2build.h: No such file or directory
```

Strange is like if it's not finding the include directory where ft2build.h is. 
I assume you are leaving platformio.ini "as is" right?
I cloned this again, my stand is 100% the same as you have in master branch. So it must be something that I have different on my toolset.
Platformio version: 1.54.2

Do you see something strange on my build output?

turgu1, 2021-03-14:
We may have found the problem... It seems that you are using an old
platformio version. It is loading an old esp-idf sdk (v4.0.1). Mine is
version 2.3.0. That  loads esp-idf v 4.2.0.  Here is the information
from the start of my own build process:
Verbose mode can be enabled via `-v, --verbose` optionCONFIGURATION: 
https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.htmlPLATFORM
: Espressif 32 (3.1.0) > Espressif ESP-WROVER-KITHARDWARE: ESP32
240MHz, 320KB RAM, 4MB FlashDEBUG: Current (ftdi) On-board (ftdi)
External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-
ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny,
tumpa)PACKAGES:  - framework-espidf 3.40200.210118 (4.2.0)  - tool-
cmake 3.16.4  - tool-esptoolpy 1.30000.201119 (3.0.0)  - tool-ninja
1.7.1  - toolchain-esp32ulp 1.22851.191205 (2.28.51)  - toolchain-
xtensa32 2.80400.210211 (8.4.0)Reading CMake configuration...LDF:
Library Dependency Finder -> http://bit.ly/configure-pio-ldfLDF Modes:
Finder ~ chain, Compatibility ~ softFound 4 compatible
librariesScanning dependencies...Dependency Graph|-- <ESP-IDF-Inkplate> 
0.9.4+sha.ef3a5f8|-- <EPub_InkPlate>|   |-- <ESP-IDF-Inkplate>
0.9.4+sha.ef3a5f8|-- <externals>|-- <tools>Building in release
modeGenerating assembly for
.pio/build/inkplate_6_release/favicon.ico.SGenerating assembly for
.pio/build/inkplate_6_release/upload_script.html.SCompiling
.pio/build/inkplate_6_release/src/controllers/app_controller.oCompiling
.pio/build/inkplate_6_release/src/controllers/book_controller.oCompilin
g
.pio/build/inkplate_6_release/src/controllers/books_dir_controller.oCom
piling
.pio/build/inkplate_6_release/src/controllers/common_actions.oCompiling
.pio/build/inkplate_6_release/src/controllers/event_mgr.oCompiling
.pio/build/inkplate_6_release/src/controllers/option_controller.oCompil
ing
.pio/build/inkplate_6_release/src/controllers/param_controller.oCompili
ng .pio/build/inkplate_6_release/src/controllers/web_server.oCompiling
.pio/build/inkplate_6_release/src/helpers/simple_db.oCompiling
.pio/build/inkplate_6_release/src/helpers/unzip.oCompiling
.pio/build/inkplate_6_release/src/main.oCompiling
.pio/build/inkplate_6_release/src/models/book_params.oCompiling
.pio/build/inkplate_6_release/src/models/books_dir.oCompiling
.pio/build/inkplate_6_release/src/models/config.oCompiling
.pio/build/inkplate_6_release/src/models/css.o

martinberlin, 2021-03-14:
Good point. I went to "PIO Home" -> Frameworks and updated Espressif 32 which updates also IDF. Now I have the same version. But still get the problem that does not find the include. Again the logs, maybe there is something I miss?
Where is the part where the folder where ft2build.h lives is referred to be included as a library?

```
Processing inkplate_6_release (platform: espressif32; board: esp-wrover-kit; framework: espidf)
-----------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32 @ ~2.80400.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa32 @ 2.80400.210211 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (1.12.4+sha.34b55a4) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40200.210118 (4.2.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 4 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP-IDF-Inkplate> 0.9.4+sha.ef3a5f8
|-- <EPub_InkPlate>
|   |-- <ESP-IDF-Inkplate> 0.9.4+sha.ef3a5f8
|-- <externals>
|-- <tools>
Building in release mode
Compiling .pio/build/inkplate_6_release/src/controllers/app_controller.o
Compiling .pio/build/inkplate_6_release/src/controllers/book_controller.o
Compiling .pio/build/inkplate_6_release/src/controllers/books_dir_controller.o
Compiling .pio/build/inkplate_6_release/src/controllers/common_actions.o
Compiling .pio/build/inkplate_6_release/src/controllers/event_mgr.o
Compiling .pio/build/inkplate_6_release/src/controllers/option_controller.o
Compiling .pio/build/inkplate_6_release/src/controllers/param_controller.o
In file included from include/models/fonts.hpp:8,
                 from include/models/css.hpp:10,
                 from include/models/epub.hpp:11,
                 from include/controllers/books_dir_controller.hpp:9,
                 from src/controllers/app_controller.cpp:8:
include/models/ttf2.hpp:11:10: fatal error: ft2build.h: No such file or directory 
```

turgu1, 2021-03-14:
Got it... I'm the culprit :-(
Here it is: I hard coded my folder location in a file that is read inside the platformio.ini file...

Look here: lib_freetype/lib/pkgconfig/freetype2.pc

The first 4 lines:

prefix=/home/turgu1/Dev/EPub-InkPlate/lib_freetype
exec_prefix=/home/turgu1/Dev/EPub-InkPlate/lib_freetype
libdir=/home/turgu1/Dev/EPub-InkPlate/lib_freetype/lib
includedir=/home/turgu1/Dev/EPub-InkPlate/lib_freetype/include

You can replace there "/home/turgu1/Dev" with the location you have installed the source code.

That was done several month ago... i missed it!

turgu1, 2021-03-14:
I will find a way to get that more generic...

martinberlin, 2021-03-14:
Yes. That was a big step forward. Maybe you can make an bash file that writes that freetype2.pc dynamically using your project directory to generate that paths (Idea)

Now I'm hitting this error:
```
*** [.pio/build/inkplate_6_release/bootloader/soc/esp32/cpu_util.o] Source `/home/martin/.platformio/packages/framework-espidf/components/soc/esp32/cpu_util.c' not found, needed by target `.pio/build/inkplate_6_release/bootloader/soc/esp32/cpu_util.o'.
```
That I'm not sure why it comes since the ESP-IDF is the right version now.

turgu1, 2021-03-14:
Here are the four lines that works. No need for a shell script:

prefix=$PROJECT_DIR/lib_freetype
exec_prefix=$PROJECT_DIR/lib_freetype
libdir=$PROJECT_DIR/lib_freetype/lib
includedir=$PROJECT_DIR/lib_freetype/include

I'm updating the project accordingly.

Please confirm that you can build the app properly.

turgu1, 2021-03-14:
Good! Thanks for your effort!

closed 2021-03-14 by turgu1

martinberlin, 2021-03-15:
@turgu1 Thanks a lot!

Quick question:  Would be OK with you if I Form this repository and make a version that supports also another epapers ?

I wanted to practice C++ and this days I don't have any client project so I though to invent my own. 
I though about analyzing what you did here and replace the rendering part to work also with my Cale-idf epaper component. 

Currently it supports only [Lilygo EPD047 parallel epaper](https://github.com/martinberlin/cale-idf/wiki/Model-parallel-ED047TC1.h) but has also like 10 other monochrome SPI epapers, that are much slower than SPI, but still can be a good target to make an eBook reader for those that do not have an Inkplate.

If it's ok with you then I will rename my fork:  EPub-epaper 
And I will give you all the corresponding Credit's since at the end 90% of the code will be yours. I will try to innovate adding things like Touch (But it consumes also some mAh per hour so battery will drain more)
Really glad to find this since it has most of the features that I wanted to achieve. Kind regards from Berlin

turgu1, 2021-03-15:
Hello Martin,
I've no problem with your project. Please do. I'm currently resolving
an issue that you can see in the github issues. Hope to resolve it
soon. I'm also working on the next version that will have a complete
CSS parser. I don't expect this version to be ready before several
weeks. There is an InkPlate-6Plus that will have a tactile display in
the futur for which I will add the functionality to interact with it.
Have fun!

martinberlin, 2021-03-15:
Ah great!  Nice to hear.
If you need some inspiration for the touch, I ported a touch component from Arduino-esp32 to ESP-IDF and currently supports:
- FocalTech FT6X36 IC controller
- L58 Another chinese touch controller that was hard to implement due to the lack of documentation

https://github.com/martinberlin/FT6X36-IDF

Nice to met you Guy!

turgu1, 2021-03-16:
Just to let you know that I corrected a bug. A new version has been released: 1.2.1
Cheers!
Guy