turgu1/EPub-InkPlate #3: Fonts looking strange on Inkplate 6

martinberlin, 2021-04-17:
Hi guy,
I'm trying this on my Inkplate 6, so far build and compiled without any problems, added some books in the SD and this is what I'm getting:
[image]
Page example:
[image]

Maybe I'm doing something wrong?
It's getting some white lines in the middle where it does not print anything.

turgu1, 2021-04-17:
Hello Martin,

From what I can see, it doesn't add blank lines but it swap even and odd lines. It could be an issue with some parameter mixed up. As you tried with a release binary, it could be a bug related to the current parameter configuration you are using. Can you send me the **config.txt** file that is located in the main folder on the uSD card please. This could help me finding the issue.

For the second issue with the release version (buttons not working), is it possible that you have used the wrong release? If you selected the one with the zip filename containing "extended_case", the touch buttons are not working as it expects the added mechanical buttons upgrade I've designed for the InkPlate-6 (look at the other project [here](https://github.com/turgu1/InkPlate-6-Extended-Case)).

martinberlin, 2021-04-17:
Sure this is my config.txt in the root of the SD card:

```
# EPub-InkPlate Config File

version = 1
wifi_ssid = "WLAN-724300"
wifi_pwd = "50238634630558382093"
http_port = 80
battery = 2
timeout = 5
font_size = 10
default_font = 1
use_fonts_in_books = 1
show_images = 1
orientation = 1
resolution = 1
show_heap = 1
show_title = 1
```
Thanks for your support!

turgu1, 2021-04-17:
In the list of releases, the third one is the one to choose for the InkPlate-6 touch buttons. I've modified the descriptive text of all releases to indicate the kind of buttons supported. 

I'll be back with my findings once I've done some tests, in the hours to come.

turgu1, 2021-04-17:
I confirm that I can reproduce the issue. It is related to the 3bits (8 gray scales) screen painting. The problem does not appears with the 1bit black/white painting. I suspect some bad pixel location computation associated to the Screen::draw_bitmap() method in the lib_esp32/EPub_InkPlate/src/screen.cpp file. I'm working on it now.

turgu1, 2021-04-17:
The issue is now corrected. I've generated 1.2.2 releases. The GitHub master source code is up to date.

closed 2021-04-17 by turgu1

martinberlin, 2021-04-17:
Amazing. I can confirm it now works correctly.
You really pushed the ESP32 to the limit with this project. Very proud to use it to read books. Thanks!!!

turgu1, 2021-04-17:
Thanks Martin, this development was a very interesting effort. I'm still working on it but at a slower pace. Next change would be a better CSS files interpreter.

martinberlin, 2021-04-17:
Great. Lot's to learn in your effort. 
I'm slowly getting to understand the 10% of it. Would love to port it with my fork so it can be used with [EPDiy parallel component](https://github.com/vroland/epdiy/) (I helped to add software rotation)
Since the recycled epaper screens of 6" are very cheap. My father found some for 6 € in spain. And then we can also make cheap epapers powered by esp32 with the potential to use any kind of parallel epaper screens (Supports many different sizes)
Very old picture of me when I was about 5:
[image]

turgu1, 2021-04-17:
Interesting. The next InkPlate 6 (the 6Plus) is going to be interesting using a touch screen and back-lit. I expect to add these features when it will be available.

martinberlin, 2021-04-17:
Back-lit that's something new for me, since most epapers I've seen are not translucent. I have a Hisense A5 epaper phone and it's not back-lit. It has front leds that illuminate the epaper from the front

turgu1, 2021-04-17:
I'm not so sure the kind of back-lit it will be. From the initial libraries published by e-Radionica, the specifications are not identified.