turgu1/EPub-InkPlate #4: Inkplate 6, continually restarting after upload

mhfowler, 2021-04-25:
I've followed the instructions, loaded the SD card, and used upload.sh which runs successfully to upload release-1.2.1.2 to my Inkplate 6. 

However, after uploading, the device just continually restarts, without ever succeeding.

I can see the device continually restarting, by opening the serial monitor in the Arduino IDE with baudrate 115200. 

Is there any way to debug further what could be the issue?

This is what I see in the monitor, repeating every second, 

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:4876
load:0x40078000,len:13032
load:0x40080400,len:4100
entry 0x40080684
```

mhfowler, 2021-04-25:
I've also confirmed the Inkplate is working when I use the Arduino example Inkplate_Basic_BW

mhfowler, 2021-04-25:
Figured out I had a 64GB sd card which was in ExFat format, 
when I changed to a 16GB in FAT32 format, then things were working :)

closed 2021-04-25 by mhfowler

turgu1, 2021-04-25:
I will modify the installation documentation to indicate that the ExFat format cannot be used.