turgu1/EPub-InkPlate #5: M5Paper Progress?

Ajb2k3, 2021-04-27:
Have you made any progress converting it to run on the M5Paper? 
I really need this please for documenting M5Products.

turgu1, 2021-04-27:
From what I can figure, the M5Paper is a discontinued device. It is no longer advertised at RoboShop, is marked as discontinued at  BangGood, and is out-of-stock at Digikey and M5Stack. Unless I can get a device to work on, I'm not going to port the software on it. If I ever get a copy of the device, the availability of a version of the EPub reader will be advertised in the M5Stack community forum. This is a project done part time so it will take sometime to get it ready as I have other ongoing activities.

Ajb2k3, 2021-04-27:
As far as I am aware, It is just out of stock due to the permanent semiconductor issues all around the world.
If there is anything I can do to help (testing wise) please ask.

turgu1, 2021-04-28:
Thanks a lot. I could certainly get you in when I will have a version ready for testing.

markbirss, 2021-05-02:
> Thanks a lot. I could certainly get you in when I will have a version ready for testing.

@turgu1 

I also have 2 of those M5Paper devices and would also be interested to see your progress with its possible future support.

Also I wanted to peek your interest in another e-paper device project - not sure if you know about it ?

https://github.com/vroland/epdiy

Im prototyping a 4.7, 6 and 9.7 inch terminal with with a Raspberry Pi 

LilyGo 4.7 (960 x 540)
6 inch WaveShare Hi-Res (1072 x 1448)
9.7 inch (1200x825)

[image]

[image]

[image]

turgu1, 2021-05-02:
This is certainly interesting. I would like to make it runs on other devices as long as they allow for interaction using tactical/mechanical buttons or a tactical screen. The InkPlate and M5Paper devices offer the required input devices. I also have designed a very simple mechanical buttons circuit for the InkPlate.

I understand that the M5Paper will get some kind of redesign by lack availability of the current eInk plate used for the current version, so it will take some time before I can get a copy. My current priority is for the close to be released InkPlate-6Plus that will have a tactical screen and infinite partial updates (no need for a slow refresh every 10/20 partial updates).

I will work on other devices as long as I can get a version of them. I'm not too much incline at designing the electronics nor going outside of the ESP32 platform (with PSRAM) as it would require a lot of changes to the software.

As software development takes a lot of time (and is free), I'm only working on donated pieces of equipment.

Cheers!

Ajb2k3, 2021-05-02:
I wish I could help out but have no spare hardware and no money to donate.
I'm glad that you wont be giving up the idea of it.

turgu1, 2021-05-02:
I don't need money really :-)
I've been developing software for the last 40 years now. It's more a matter of having the (part) time to do it and the appropriate hardware to develop, test and support when there is issues to solve.

turgu1, 2025-03-16:
Hello Domigome! I got a M5Paper a year ago. I tried to upload a very simple code to it, written with esp-idf (just saying “hello” on the usb port). It bricked the device! Tried for several days to get it back running, reinstalling the original code in it without success. Tried to get some help from the community without success. It’s gone for me.

turgu1, 2025-03-16:
I'm going to try to recover it again. One of my friend gave me some hints to find the issue (could be related to a hardware problem as the schematics shown that it must be working well with ESP-IDF). Will open it and put my oscilloscope in duty...

turgu1, 2025-03-19:
Found the problem: No power to the device. Did changed some diodes and two transistors require to be changed. Waiting to receive the transistors (mosfet kind I believe). My friend engineer helped me diagnose the issue. Hope that this will permit me to restart the project sometimes during next summer.

turgu1, 2025-03-26:
Some update: my friend engineer resolved my issue with the device. The battery was also replaced. The device seems to be in a working condition now. I've been able to program the classes required to read the battery voltage level, the buttons and the touch screen. Still working on the display class to get it runs now. I don't expect to have a version of the EPUB reader until spring or summer time frame, as I have other duties to take care of. Will work on it from time to time.

closed 2025-03-31 by turgu1