turgu1/EPub-InkPlate #7: Partial Refreshs of EPD

DerNuntius, 2021-09-10:
Thank you for writing this great software for the inkplate! It's really impressive that an esp32 can leverage such work. Is it possible to implement partial refreshs to save battery life and speed up refreshs?

turgu1, 2021-09-11:
Hello Hugo,

Thanks for your request.

There are two refresh activities in the application:

- The first one is the refresh of available ebooks on the SDCard. This one is done every time the ESP32 is wakening up after deep sleep or at cold-boot time. It is very easy for me to add an entry in the main parameter menu that would allow for manual refresh instead of every time the device is restarted. I can effectively implement it in the next release (v1.3.0) that I expect to get out in a week or so.

- The second one is the refresh of pages location in a book. This is triggered when a book is opened for the first time or when a display parameter is changed that would have an impact on pages formatting. That one is complex and if done partially, wouldn't allow the display of page numbers and (for the new upcoming version) wouldn't permit access to the table of content for some of the books that require the book scan to find the chapters. If someone doesn't play with formatting parameters (and landscape/portrait orientation) that refresh would be done only once. I understand the issue with the battery life. I will have to analyze how I can implement something,  considering that a user would want to get control over it, notwithstanding the loss of functionality with page numbering and the table of content. Don't expect to get that one in version 1.3.0, unless it is trivial to do...

Hope this helps.
Guy

DerNuntius, 2021-09-11:
Thank you for the feedback.

What I meant was the partial refreshing of the epaper display. Instead of a full refresh cycle for every page it's faster to partially refresh the display. The kindle does this to reduce the time it take to draw a new page.

turgu1, 2021-09-11:
It’s already there! 

When you select 1bit pixel resolution instead of 3bits in the Main Parameters, you get partial refresh. This is explained in the user’s guide.

DerNuntius, 2021-09-11:
Ooops sorry for the inconvenience :)

turgu1, 2021-09-11:
No problemo!

It allowed me to think about other potential enhancements (as long as they are efficient at saving some battery power). I won't implement it for the upcoming release but may appear in the future.

closed 2021-09-11 by turgu1