turgu1/EPub-InkPlate #8: Dynamically add/remove default fonts?

loglow, 2021-09-25:
Hello, I'm wondering if it would be possible to dynamically scan the `fonts` directory so that new fonts can be added without changing the code or reflashing the device? I see that currently the selection of fonts is hard-coded in `src/models/fonts.cpp` and it would be nice to be able to add new default fonts just by dragging them in. I think requiring the same naming convention ("FontName-Regular", "FontName-Bold", "FontName-Italic", "FontName-BoldItalic") would be fine. I assume that any otf or ttf fonts would already be supported by the software. Thanks and let me know what you think!

turgu1, 2021-09-25:
Hello loglow, that is something I was thinking to add in the next release. It will be a table with up-to-height entries that will appear in the application forms, replacing the default fonts list. I discovered a bug in version 1.3.0 that I've corrected today to be released in v1.3.1 and will look at adding this feature also. I will have to limit the size of those fonts to 50Kbytes as the available memory wouldn't allow using too big font files.

loglow, 2021-09-25:
Cool, sounds interesting! Looking forward to trying it out.

turgu1, 2021-09-26:
Release 1.3.1 is now available. Look at the user's guide, section 3.7 for instructions on how to use the feature.
Cheers!

closed 2021-09-26 by turgu1