turgu1/EPub-InkPlate #9: Thoughts about pixel-perfect fonts...

loglow, 2021-09-26:
I've been playing around with different fonts, and here are some of my thoughts...

I have an Inkplate 10, and the DPI isn't incredibly high (approximately 146 DPI, if my math is right), so unfortunately, fonts don't really look all that great. This is because the built-in vector fonts are intended for use with very high DPI displays (like phones) or full anti-aliasing. On the Inkplate, they look sort of choppy, which just isn't great for long-term reading. (I do realize that fonts _are_ somewhat anti-aliased in 3-bit mode, but I strongly dislike 3-bit mode because a) text often appears very washed-out, b) it still looks pretty choppy, and c) I greatly appreciate the partial refreshes offered in 1-bit mode).

The solution to this, of course, is to use bitmap (aka raster) fonts at their native resolutions. For example, use of the `PPMondwest-Regular.otf` font (attached) with the size set to "12" looks very clean. It does not look clean at other sizes (at this time the other size options are "8", "10", or "15"). I put these numbers in quotes because they're what's used in the settings, and they are **pt** values, which are more confusing when talking about bitmap fonts. (`PPMondwest-Regular.otf` is a 16 **px** font, which is perfectly equal to 12 **pt**.)

The hard part is a) finding good bitmap fonts, b) at varying sizes, and c) with full bold/italic variants. However, I think it would be a worthwhile effort, because text just looks **so much better** with pixel-perfect fonts. Perhaps old computer operating system fonts would be a good source, since 2-3 decades ago basically all fonts were bitmap? Perhaps there are modern sources for good bitmap font families? I'm not sure.

Let me know what you think!

Attachment: [PPMondwest-Regular.otf.zip](https://github.com/turgu1/EPub-InkPlate/files/7230516/PPMondwest-Regular.otf.zip)

turgu1, 2021-09-26:
Hello again :-)

Welcome to the complexity of fonts selection for low-resolution devices! I was used to those old fonts 35 years ago :-)

This could certainly be an option, as long as we can find some fonts that would be good at the other sizes. One issue would be to keep the font file size low enough to be usable with the application.

turgu1, 2021-09-26:
Here is a place on the web to look at bitmap fonts:

http://www.fontsaddict.com/font/category/bitmap

Not sure if something there would be appropriate.

turgu1, 2021-09-26:
Now that version 1.3.1 is out, it is possible to do some trials with other fonts. I have done several trials myself and it is not obvious how to find fonts that would be suitable for the output quality AND the limitations imposed by the device... I will continue to try to find a solution on my side. There is still space to add another font display algorithm in the application so if we can find something even outside of the TTF/OTF formats, there may be enough room to add something.

You may certainly find a way to fit English-only fonts, but this is rather limiting. I would prefer fonts that support latin+ characters. Here is the list of the characters included (when available) in the current application fonts (Unicode codification):

U+0000-00FF,U+0131,U+0152-0153,U+02BB-02BC,U+02C6,U+02DA,U+02DC,
U+2000-206F,U+2074,U+20AC,U+2122,U+2191,U+2193,U+2212,U+2215,
U+FEFF,U+FFFD

Let me know if you find anything worthwhile.

I'm closing it now but revisit the issue sometimes.

closed 2021-09-26 by turgu1

turgu1, 2021-09-26:
I'm just thinking about a suite of tools that were specifically developed to generate bitmap fonts: METAFONT is part of the TeX suite of tools. Some info is available here: https://texfaq.org/FAQ-useMF

You can define the parameters for the output resolution and point sizes. You can also control which characters to output. There is not a large variety of fonts available, but the output is generally optimal for the resolution required.

The TeX suite needs to be installed to use Metafont. The resulting fonts are not TTF but their format is well documented and a driver could be built... I will try to play with it. Last time I used it was in 1986...

turgu1, 2021-09-27:
I've been looking at the PPmondwest font. This is not a true bitmap font but a mimic of a bitmap font using the usual vectorized font. A 16px font is equivalent to 12pt size font on a 96 dpi screen (16*72/96). To get a 12pt size font on 150 dpi screen would require a 25px font.

I've used FontForge with your font and generated 31px (15pt), 25px (12pt),  21px (10pt), and 17px (8pt) bitmaps with it. 

you can compare the '@' representation in the attached pictures. You can see that the result is not that bad. The green lines show the outline of the vectorized version, on top of the grey pixels generated using the FreeType algorithm. I will make a try of your font with a downsized version of it on my Inkplate-10.

[image]
[image]
[image]
[image]
.

turgu1, 2021-09-27:
Well, I realize that with such a font, you don't get a better resolution as you augment the size of the glyphs. When you look at the font at a very high resolution, you only get a mimic of a bitmap font.