turgu1/EPub-InkPlate #12: Wifi Stack Error: wifi:Group & Pairwise ciphers must be AES-CCM for PMF

hmelder, 2022-02-20:
Connecting to the AP fails repeatedly with the following error: "wifi:Group & Pairwise ciphers must be AES-CCM for PMF"
[image]

turgu1, 2022-02-20:
How is your wifi network configured for security? (WPA2, WPA3, RADIUS, PMF ? Other parameters beyond ssid and password?)

hmelder, 2022-02-20:
I've checked the preferences of the AP and sure enough PMF was enabled. Thank you for the quick response :)

closed 2022-02-20 by hmelder

turgu1, 2022-02-20:
Which security protocol are-you using, is it a router, what model?

turgu1, 2022-02-20:
It may be that the PMF on your AP is using an opmode different than AES-CMM (like AES-GCM, CNSA, SAE-AES or others). Please try to find which one it is using.

hmelder, 2022-02-20:
It works now. My router is a Vodafone EasyBox 804.