turgu1/EPub-InkPlate #13: Pagination Issues for Dual Foldable InkPlate 6Plus Reasers

feebb, 2022-03-17:
Dear Guy,

Many, many thanks for your excellent work on the Inkplate ePub Reader!

Since I love the idea of having an eReader that feels and acts more like a real book, I have put together a rudimentary folding ebook reader using two Inkplate 6Plus readers and a 3D Printed Case (The case currently only folds 90 degrees, but that is enough for now for the purpose).
[image]
 I know this is a poor substitute for a foldable display, but eInk doesn't seem to want to treat this as a priority. 
[https://www.engadget.com/e-ink-folding-ereader-prototype-170116617.html](url)

The problem of course with this solution is page turning. Reading the same book at the same time with 2 eReaders means 2 separate page turns for each device to advance the book properly in both eReaders.

Based on my rudimentary knowledge of programming and your excellent explanations in your readme and User Guide, I understand that pagination is a calculation involving the size of the screen, the font size, a variety of other factors that vary from book to book, etc.

What I was wondering was whether it would take a lot of coding work to implement one or both of the following:

(1) Make one tap of each ereader advance the book 2 pages (thereby eliminating two of the four taps currently needed); and

(2) Make it possible to have the option to change the direction of the page flow with the side of the screen being tapped--in other words, make it possible to go forward with a tap on either the right or left side, assigning backwards to the remaining side.

I realize that while the second idea might have some general application, the first idea would only make sense for the dual ereader scenario.  Unfortunately, I have no idea how to implement either of these ideas. What i was wondering was whether, if the changes did not require a lot of coding for you, you could direct me to the proper place in the source code for me to plug it in, and then recompile the software for my machines only.

I know this is a lot to ask for such a limited use, but any help at all would be greatly appreciated!

Thanks again for all you've done on this project.

Best,

Rod Stenlake

P.S. I would have posted this on the eRadonica message board, but I have been unable to log in for quite some time.

turgu1, 2022-03-17:
Hello Rod,

First, thank you for your kind words! This seems to me to be a very interesting idea!

The best approach would certainly be using one of the two devices as the **master** and the other the **slave**, such that only the master device would have the SDCard with the documents, and, once a book is open, it is automatically copied to the slave device that will synchronize from receiving master commands and vice-versa (maybe...). A lot of work would be required to implement something like that...

Your approach is easier. I would add a configuration parameter to indicate if the device is showing only even or odd pages, another parameter to inverse the next/previous page touch location on each device. As the devil is in the detail, I would have to analyze the impact of such a change in the page generation process... other functions may be required (e.g. what to do when synchronization is lost?).

A not so difficult feature to add would be to connect both devices through the EasyC interface and send the command to the other device when a change of page is requested, so, only one touch on the screen would impact both devices. As I'm not that much familiar with that interface, I presume that this is doable. Using BlueTooth to do that would also be an option, but will impact both the available power and the required memory to add this protocol.

I presume here that you are using the Inkplate-6Plus that uses a touch screen. The same idea would be possible with touch buttons or mechanical buttons. Please let me know which device kind you are using.

I'm a bit busy this month but could look at this feature for the next release that I expect to get out within a 2 or 3-month timeframe.

As I don't have two devices of the same kind, you may become my Guinea pig for trials :-)

Cheers!

Guy

ThorstenChalas, 2023-11-01:
This feature request would also be interesting for me to build a real looking book for live action roleplaying. Would be a good magic book if you Bild a nice in-game cover. For such a project I would try to get a bit into modification of the UI to make it look less technical and maybe even adding a function to start books by RFID empedef in a epoxy Chrystal or so but that would be added later. Having the option to use dual screens like described above would be really handy to make the long story short. Really great work what you have already achieved