turgu1/EPub-InkPlate #14: /usr/bin/pkg-config: No such file or directory  (M1 Mac/InkPlate 10)

nwfisk, 2022-11-11:
So, I've tried just about everything to debug this issue over the past week, but I simply cannot seem to get this project to build correctly. I get the feeling this is a path issue, which I've tried to correct. 

I will mention that I've struggled with the readme somewhat, insofar as it's not always clear which steps are required and which steps have already been completed - so I'm hoping I read correctly and set up my environment correctly.

Any suggestions on how to start debugging this would be greatly appreciated!

`Processing inkplate_10_release (platform: espressif32; board: esp-wrover-kit; framework: espidf)
------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp-wrover-kit.html
PLATFORM: Espressif 32 (5.2.0) > Espressif ESP-WROVER-KIT
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (ftdi) On-board (ftdi) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.40402.0 (4.4.2) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
/bin/sh: /usr/bin/pkg-config: No such file or directory
OSError: '/usr/bin/pkg-config --cflags --libs lib_freetype/lib/pkgconfig/freetype2.pc' exited 127:
  File "/Users/nate/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py", line 188:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/nate/.platformio/platforms/espressif32/builder/main.py", line 312:
    target_elf = env.BuildProgram()
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/nate/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/nate/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 117:
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/nate/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 217:
    env.Append(**env.ParseFlagsExtended(flags))
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Util.py", line 737:
    return self.method(*nargs, **kwargs)
  File "/Users/nate/.platformio/penv/lib/python3.10/site-packages/platformio/builder/tools/piobuild.py", line 180:
    for key, value in env.ParseFlags(str(raw)).items():
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Environment.py", line 834:
    do_parse(arg)
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Environment.py", line 686:
    arg = self.backtick(arg[1:])
  File "/Users/nate/.platformio/packages/tool-scons/scons-local-4.4.0/SCons/Environment.py", line 597:
    raise OSError("'%s' exited %d" % (command, status))
==================================================== [FAILED] Took 0.23 seconds ====================================================

Environment          Status    Duration
-------------------  --------  ------------
inkplate_10_release  FAILED    00:00:00.234
============================================== 1 failed, 0 succeeded in 00:00:00.234 ==============================================`

turgu1, 2022-11-12:
Hello,
At this point in time, I'm using the Espressif platform version 5.1.0.

It seems that you don't have the pkg-config tool that is supplying the proper include parameters for FreeType. Don't know if it is available for MacOS. 

In platformio.ini, you may try to replace line 42:

```
  !/usr/bin/pkg-config --cflags --libs lib_freetype/lib/pkgconfig/freetype2.pc
```

with

```
-I$PROJECT_DIR/lib_freetype/include/freetype2 -L$PROJECT_DIR/lib_freetype/lib -lfreetype
```

Normally, the only external dependency is the ESP-IDF-Inkplate library which is automatically retrieved from my GitHub project. All other dependencies (other than ESP-IDF itself) are in the source tree.

nwfisk, 2022-11-13:
Excellent - this worked, thank you!

turgu1, 2022-11-13:
You are welcome!

closed 2022-11-13 by turgu1