turgu1/EPub-InkPlate #15: Feature Request: Button to Force Page Refresh

BlipRanger, 2023-01-21:
I was thinking it might be nice if there was a menu option to force an immediate page refresh, or at least modify how often it occurs. I might consider trying to implement this myself, but I'm really unfamiliar with the code/libraries involved, so it may take some time.

turgu1, 2023-01-22:
That is a good idea. I expect to have a new version ready around mid of February. I may limit the maximum delay for a full refresh to 15 pages, as it could damage the screen.

BlipRanger, 2023-03-17:
> That is a good idea. I expect to have a new version ready around mid of February. I may limit the maximum delay for a full refresh to 15 pages, as it could damage the screen.

Awesome, thanks, looking forward to it!