turgu1/EPub-InkPlate #18: Inkplate 6 PLUS boot loop

gigawatts, 2023-06-18:
I can't get either the 2.0.0 or 2.0.1 release for the Inkplate 6 Plus to start. This is using the "Soldered" variant of the device (https://soldered.com/product/soldered-inkplate-6plus-with-enclosure-e-paper-display-with-touchscreen-copy/), I am not sure if that makes a difference.

Here is the boot sequence I get, followed by a reboot, over and over, with or without an SD card inserted.

```
I (721) psram: This chip is ESP32-D0WD
I (722) spiram: Found 64MBit SPI RAM device
I (723) spiram: SPI RAM mode: flash 40m sram 40m
I (724) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (731) cpu_start: Pro cpu up.
I (735) cpu_start: Starting app cpu, entry point is 0x40083c64
I (0) cpu_start: App cpu up.
I (1626) spiram: SPI SRAM memory test OK
I (1635) cpu_start: Pro cpu start user code
I (1635) cpu_start: cpu freq: 240000000
I (1636) cpu_start: Application information:
I (1638) cpu_start: Project name:     EPub-InkPlate
I (1644) cpu_start: App version:      2.0.1
I (1649) cpu_start: Compile time:     May  1 2022 18:56:00
I (1655) cpu_start: ELF file SHA256:  1f56aaecd8043574...
I (1661) cpu_start: ESP-IDF:          4.3.2
I (1666) heap_init: Initializing. RAM available for dynamic allocation:
I (1673) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1679) heap_init: At 3FFC13C0 len 0001EC40 (123 KiB): DRAM
I (1686) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1692) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1699) heap_init: At 4009FDD0 len 00000230 (0 KiB): IRAM
I (1705) spiram: Adding pool of 4084K of external SPI memory to heap allocator
I (1714) spi_flash: detected chip: generic
I (1718) spi_flash: flash io: dio
I (1729) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1732) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 3, 4MB external flash
Minimum free heap size: 4373811 bytes
Heap summary for capabilities 0x00000C06:
  Totals:
    free 0 allocated 0 min_free 0 largest_free_block 0
I (1762) main: EPub-Inkplate Startup.
I (1782) MCP23017: MCP at address 0x20 has been detected
I (1792) MCP23017: MCP at address 0x22 has NOT been detected
E (1792) MCP23017: The MCP at address 0x22 has not been detected.
E (1792) MCP23017: The MCP at address 0x22 has not been detected.
I (1932) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:2 
E (3982) Wire: Unable to complete request_from: ESP_ERR_TIMEOUT.
E (3982) main: InkPlate6Ctrl Error.
E (3982) main: Config Error.
E (3982) Fonts: Unable to read fonts definition file fonts_list.xml.
E (3992) main: Font loading error.
E (3992) Fonts: Fonts.get(): Wrong index: 0 vs size: 0

abort() was called at PC 0x401ec55a on core 0

Backtrace:0x400838de:0x3ffdd9b0 0x4009255d:0x3ffdd9d0 0x4009a32a:0x3ffdd9f0 0x401ec55a:0x3ffdda60 0x401ec587:0x3ffdda80 0x401e9ddf:0x3ffddaa0 0x401d58bb:0x3ffddac0 0x400f8157:0x3ffddd60 00

ELF file SHA256: 1f56aaecd8043574

Rebooting...
```

I have tested that the SD card can be read from, with the MicroPython firmware and this example: https://github.com/SolderedElectronics/Inkplate-micropython/blob/master/Examples/Soldered_Inkplate6PLUS/exampleSd.py

gigawatts, 2023-07-14:
I asked the Soldered Electronics Inkplate support folks about the hardware differences, here is their reply:

> yes, there is a difference in the hardware and pin connections. See this link for some useful info - the main difference is the GPIO expanders: https://inkplate.readthedocs.io/en/latest/arduino.html?highlight=mcp#io-expander-functions
>  
> There would have to be some modifications in the code for it to run on 'Soldered' variants of the product. They have to do with the GPIO expander and how the driver communicates with the ePaper via the GPIO expander - this is also one of the reasons used a different board definition.
>  
> I just had a look at that repository expecting our official version of the Inkplate library to be included somewhere (searched for #include "inkplate.h") but I can't seem to find that, so this user wrote their own driver for the ePaper and just used our hardware. So, it seems to me you're gonna have to see with them is this a feature which they can implement. 
>  
> I have done this adaptation from MCP to PCAL for a few boards in Arduino within our native drivers and in micropython and didn't have too many problems. My best advice would be to find any place where the GPIO expander is used to drive the screen and change the functions accordingly. Also, check the schematics for Soldered Inkplate 6PLUS in the hardware repository: [https://github.com/SolderedElectronics/Soldered-Inkplate-6PLUS-hardware-design ](https://github.com/SolderedElectronics/Soldered-Inkplate-6PLUS-hardware-design)You can also use our arduino driver as a reference.

I may take a crack at porting this, but figured I would post their reply here, as you can likely adapt the changes much more efficiently than I can.

gigawatts, 2024-06-06:
I'm attempting to port support for this PCAL GPIO expander in myself, but I can't find a combination of versions between any ESP-IDF-InkPlate library branches/versions, this repo's branches, and platformio/framework-espidf versions that compiles as is without a massive list of errors.

What combinations are supposed to work as is?

turgu1, 2024-06-06:
Hello, sorry for the delay to look at this issue as I’m very busy under contract. 

I will look at your issue in the next days. I may generate a new version of the code, replacing de MCP with the PCAL. This is for the Inkplate 6Plus?

turgu1, 2024-06-06:
As I don’t have rhe hardware, you may have to be my tester!

gigawatts, 2024-06-06:
Yeah, 6 Plus, from Soldered. Happy to test anything!

gigawatts, 2024-06-17:
@turgu1 any progress and anything I can help test?

turgu1, 2024-06-17:
Hello! Not yet I'm afraid. I'm porting it to the last version of the ESP-IDF and got some issues. Hope to have something soon. You will be the first to ear about it.

swlxe, 2024-07-20:
@gigawatts @turgu1 Has there been any update on this issue? I'm having the same issue and also bought this device just for this use case...

turgu1, 2024-07-20:
Hello, I’m still working under contract so the time I can take on this is very limited. As I didn’t touch this software for the last 2 years is also requesting more time to regain the insights of how things were done.

I found my issue with the last esp-idf release. I have a portion of the new PCAL driver completed (30%). Some changes in the use of the i/o port request some other changes to the code (enabling the SD card at boot time) and maybe other things as I need to check the changes between 6PLUS v1 and v2 schematics. Hope to complete a first cut in the next 2 or 3 weeks.

gigawatts, 2024-07-21:
Sounds good, my offer still stands on any beta testing you need performed on hardware!

swlxe, 2024-07-21:
I'm also willing to beta test if you need another! I look forward to any progress.

turgu1, 2024-07-26:
I'm almost there. In both ESP-IDF-INKPLATE and EPUB-INKPLATE projects you will find a branch named 6plusv2-integration that contain the current updates done, required to support the PCAL6416 and the creation of the Inkplate 6Plusv2 app.

THIS IS NOT READY YET! Still remains some adjustments to the code on both side. Can't work on it for the next 2-3 days, but we are close to be able to compile and do some testing.

Before integrating these branches to the main one, I will have to test with the other MCP23017-based devices as the changes may have an impact on their stability.

So stay tuned!

gigawatts, 2024-07-27:
Awesome! Looking forward to testing!

swlxe, 2024-07-29:
Great, I look forward to more updates and also hopefully some testing.

turgu1, 2024-08-11:
Some news: modifications completed at 99%, will update github tonight. Here are the issues I still have to resolve:

- I was targeting ESP-IDF V5.3. The IRAM0 is being overflown of 10KBytes. Will look at going back to a previous versions as I have tried to resolve the issue without success yet.
- Double check is ongoing with the Soldered Arduino library source code to validad all changes needed in support of the v2 devices.
- Got a hardware problem that was corrected last week.
- Still need to get some testing with the current 6Plus before giving you a version for testing.

turgu1, 2024-08-12:
Thanks for the offer. I never asked any money for this project and all other projects available in my account on GitHub. This is my way of returning to the developers’ community for everything that is made freely available on the net, that I was able to access for the last 40 years.

gigawatts, 2024-08-20:
As always, thanks for the update! 

Interesting, is the Flick the replacement for the battery powered 6 Plus? They seem to have similar specifications.

turgu1, 2024-08-20:
I don't know enough about the Flick yet. I understand that it is very close to the 6PLUS V2. It uses a different touch layer on top of the display and used the PCAL chip instead of the MCP as per the V2. I've seen some added filtration capacitors in the schematic. Still a lot of analysis to do...

BTW the PCAL driver is able to be compiled now. Remains to do a double check comparing my code with the Solderer's Arduino library.

turgu1, 2024-08-28:
Some new… 

- I caught de flu a week ago so nothing really advanced since my last message. Still recovering.
- I received the Inkplate 6Plus FLICK. This will permit me to have the PCAL support validated.

swlxe, 2024-09-26:
@turgu1 Any updates? Anything you need from us?

turgu1, 2024-10-01:
Hello, The month of September was awful for me (it was not a flu, but Covid, I also discovered an issue with my house that requires to be repaired before winter). Still another week that I will not have time to continue the effort. I’m a bit frustrated with this but no choice…

swlxe, 2024-10-02:
Sorry to hear you got sick again. Ive battled the same issue last year or so with getting covid back to back to back. Hope you are recovering well. And of course put your health and home first. As always thanks for the work and the update!!!

turgu1, 2024-10-14:
Hello, I'm please to inform you that a first BETA version for the Soldered Inkplate-6PLUS (V2) is now available at the following location: https://github.com/turgu1/EPub-InkPlate/releases/tag/v2.1-BETA.1-Inkplate-6PLUSV2

Please do some trials and let me know if you get any issue with it.

gigawatts, 2024-10-14:
Fantastic! I'll give this a whirl tonight, thank you!

turgu1, 2024-10-14:
I've just updated the release, so if you already downloaded it, reload again. A bug found in 3Bit display mode (grayscale) that render  the menus as a black rectangle.

gigawatts, 2024-10-14:
> I've just updated the release, so if you already downloaded it, reload again. A bug found in 3Bit display mode (grayscale) that render the menus as a black rectangle.

I just downloaded the release zip from https://github.com/turgu1/EPub-InkPlate/releases/tag/v2.1-BETA.1-Inkplate-6PLUSV2  and it says updated "3 hours ago" but it seems to still have the black menus in 3bit mode.

turgu1, 2024-10-15:
Hum... I will check.

turgu1, 2024-10-15:
I may have missed to save a file. I'm rebuilding the application now.

turgu1, 2024-10-15:
Update completed. Please look here: https://github.com/turgu1/EPub-InkPlate/releases/tag/v2.1-BETA.2-Inkplate-6PLUSV2

gigawatts, 2024-10-15:
Tried the new release, same effect for me, black menus in 3bit mode

turgu1, 2024-10-15:
Strange. I would have expected that it would behave the same as per the e-radionica 6PLUS for the display, as it is using the same initialization code. My 6PLUS behave properly here.

OK, I will revert the color to what it was... stay tune for BETA.3 in a moment...

turgu1, 2024-10-15:
Here it is: https://github.com/turgu1/EPub-InkPlate/releases/tag/v2.1-BETA.3-Inkplate-6PLUSV2

I will need to investigate as I still don' t understand that...

turgu1, 2024-10-15:
Please try the BETA.3 version to see if it resolves the issue. Still can't find any specific info about it.

gigawatts, 2024-10-15:
I tried it last night, didn't see any real difference between any of the beta releases regarding the 3 bit mode.

turgu1, 2024-10-15:
I found the problem... a #define named `INKPLATE_6PLUS_V2` was misnamed `INKPLATE_6PLUSV2`.

Here is the new version: https://github.com/turgu1/EPub-InkPlate/releases/tag/v2.1-BETA.4-Inkplate-6PLUSV2

gigawatts, 2024-10-15:
Nice, that fixed the 3bit display mode!

Everything seems to be working, though I do see these warnings on the serial console during boot (indicated with arrows at the ends of the lines:

```
I (1639) esp_psram: Adding pool of 4081K of PSRAM memory to heap allocator
I (1648) spi_flash: detected chip: generic
I (1651) spi_flash: flash io: dio
W (1658) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`  <----------------
I (1669) main_task: Started on CPU0
I (1679) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1679) main_task: Calling app_main()
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 300, 4MB external flash
Minimum free heap size: 4381200 bytes
Heap summary for capabilities 0x00000C06:
  Totals:
    free 0 allocated 0 min_free 0 largest_free_block 0
I (1699) main: EPub-Inkplate Startup.
I (1719) PCAL6416: MCP at address 0x20 has been detected
I (1719) main_task: Returned from app_main()
I (1729) PCAL6416: MCP at address 0x21 has been detected
I (1819) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:2 
I (2099) TouchScreen: Resolution: [1436, 1920]
I (2109) SDCard: Setup SD card
I (2159) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2199) sdspi_transaction: cmd=5, R1 response: command not supported  <--------------
Name: SU02G
Type: SDSC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 1886MB
CSD: ver=1, sector_size=512, capacity=3862528 read_bl_len=10
SSR: bus_width=1
```

turgu1, 2024-10-15:
These issues are already part of my current todo list before releasing the official versions. Please let me know if you find any other problem. I will leave this opened for further updates.

ToDo list:
- too many interrupts from touchscreen may cause reboot
- watchdog trigger not enough when long computation occurs
- Remove some menu refresh for 3bit mode
- Update the i2c driver
- Checks on sd card busy state before deep sleep / light sleep
- command not supported error for sd card
- Update the freetype library
- Update the user guide about the danger of corrupting the sd card at power off.
- readme of both epub-inkplare and esp-idf-inkplate updates 
- CMakefiles.txt update for all esp-idf-inkplate examples

gigawatts, 2024-10-15:
Awesome! Thanks for the great work! I'm going to load up a few books and give it a deeper "real use" test.

I have found a few spelling mistakes, both in the docs and in some of the firmware menus/text boxes. I can PR some of those, though I'd like to try compiling the firmware myself first before I create any firmware based PRs, just so I know they work. I'll likely wait for your updated build instructions first.

One small feature request is to allow the DNS hostname that the firmware sets when joining a network to be configured via `config.txt` on the SD card. Right now it defaults to "espressif". This is also likely something I could PR if I had build setup instructions.

A bigger feature request I could see being useful is an "Access Point" WiFi mode, where the esp32 itself generates a AP that you could connect to with a mobile device for easy "away from home" loading of new books, without fiddling with an SD card reader in the field.

turgu1, 2024-10-16:
Interesting ideas, here are my answers:

- spelling mistakes: please do, english is my second language that I do not control completely. Thanks for the offer!
- DNS name: no problemo I will look at doing it properly, adding a config entry for it, done it already for another project
- AP mode: this can be done as long as this is not required all the time. WiFi and the web server are taking too much memory to have them running at the same time a book is being read. Could be a second way of launching the device’s web server capability. I will look at it once the rest of my todo is completed.

swlxe, 2024-10-16:
I'm stuck in a boot loop. I'm gonna backtrack and try to flash again. 

[image]

swlxe, 2024-10-16:
I'm seeing a line through my screen on the left side and show shadowing. Not the end of the world, may also just be my device...

[image]

turgu1, 2024-10-17:
Hope that your device is not having a hardware issue. It seems to me that you don't have a battery installed as it shows 0 volts at the bottom, is it?

turgu1, 2024-10-17:
I have a e-radionica 6PLUS, not a 6PLUSV2 (The V2 is for Soldered versions). The screen electronic is the same. I don't have a battery installed and I don't get this bar. Could be a good idea to submit the issue to the Soldered guys. They may have an idea to resolve that.

turgu1, 2024-10-17:
Can you try to upload one of the Arduino applications to see if the bar disappear? If so, there maybe something I can do, adding some additional delay in the screen refresh algorithms.

swlxe, 2024-10-17:
The issue seems to fix itself when I use 3bit resolution the problem disappears.

turgu1, 2024-10-17:
OK So it must be related to the 1Bit resolution screen update algorithms. Will check that.

turgu1, 2024-10-17:
The cmos battery coins is just for the RTC.

turgu1, 2024-10-17:
There exists some code in the Arduino library that cope with the lack of a battery. Will check to add it to my code.

swlxe, 2024-10-17:
I is its not a feature yet, Id like to request a feature to manually set the page. Doesnt need to be on the device itself just somewhere.

turgu1, 2024-10-17:
I'm not sure I understand you request. The table of content is not allowing to be close enough to the page you want?

swlxe, 2024-10-18:
No, you know what that may be a pdf to epub formatting issue... Let me see if I can fix it manually.

turgu1, 2024-10-21:
Greetings, here is just an update to where I'm now. It was a long week of work, mainly related to to the I2C interface changes as the legacy one will soon be unavailable in the ESP-IDF library. I will push another BETA version in a day or two once I can finish some testing. Here is my ToDo update:

```
ToDo list
---------

Last update: October, 20th 2024

[X] too many interrupts from touchscreen may cause reboot
[X] watchdog trigger not enough when long computation occurs --> watchdog disabled
[ ] Remove some menu refresh for 3bit mode
[X] Update the i2c driver (using i2c_master.h) --> major update!
[-] Checks on sd card busy state before deep sleep / light sleep --> not possible
[X] command not supported error for sd card --> OK. not error, but info
[X] mDNS support for Web Server
[X] front light i2c comm failed issue
[X] touchscreen i2c bug at boot time
[ ] add battery-less support --> not for EPub-Inkplate
[ ] AP Mode for Web Server
[ ] e-Radionica Inkplate 6 update and tests
[ ] e-Radionica Inkplate 10 update and tests
[ ] keys extension to build system
[ ] readme of both epub-inkplare and esp-idf-inkplate updates
[ ] CMakefiles.txt update for all esp-idf-inkplate examples
[ ] Update the freetype library
[ ] Update the user guide about the danger of corrupting the sd card at power off.
[ ] Update the user guide on new config parameters
```

gigawatts, 2024-10-21:
I've been reading a book from mine for the last few days, in 3bit mode and it's been working great!

One question though, during every power on sequence, it displays the note about "new ebooks have been found", despite no new files having been added. Is this normal?

gigawatts, 2024-10-21:
Oh and another suggestion. From what I can tell, the "WAKE" button serves no purpose when the device is awake. Can it be configured to place the device into deep sleep mode manually (instead of waiting for the sleep timeout) and maybe even disable the power circuit as well? 

It's easy to forget that I need to put the device into deep sleep and then also hit the I/O button to kill power. Just thinking about user experience enhancements, nice to haves :)

turgu1, 2024-10-21:
@gigawatts : 

- for the  "new ebooks have been found" this could be a sign that the sd-card content is corrupted. Maybe a good idea to do a repair on a Windows based computer
- for the "wake button", this is a good idea. Will check if I can get access to it and implement a manual deep-sleep.
- for the off button, I will look at it's potential of control. If I can read it and disable it, it could also be used to enter deep-sleep.

turgu1, 2024-10-21:
@gigawatts : My first look at the schematic doesn't permit me to see how I can control the on/off switch through software. You can add a resistor on R34 to disable it, but it's an hardware modification...

For the wake button, will add an entry in my todo :-)

turgu1, 2024-10-21:
@gigawatts well got a similar issue with the wake up button: it is only hooked to the gpio 36 that is the interrupt pin used for both waking up and all touch screen events. Can’t find a way to infer when the interrupt occurs, if it is from the wakeup pin being pushed and not the touch screen. Still investigating… a bit sad as it is a very good idea.

turgu1, 2024-10-25:
@gigawatts One way to cause a manual deep sleep is to use the power-off menu entry available both in the main menu and in the book reading menu.

gigawatts, 2024-10-26:
@turgu1 yeah, that's what I've been using, was just thinking it would be nice to be able to trigger that quickly from a hardware button.

turgu1, 2024-10-28:
A new BETA version is now available:

https://github.com/turgu1/EPub-InkPlate/releases/tag/v2.1-BETA.5-Inkplate-6PLUSV2

[4 more replies]