turgu1/EPub-InkPlate #22: How to build 6plusv2-integration branch?

silbe, 2025-06-22:
I'd like to give adding InkPlate 10 v2 (MCP23017 replaced by PCAL6416) support a try now that support for 6PLUS v2 (also using PCAL6416) has been added. As a first step I'm trying to build the 6PLUS v2 release from source (without changes for 10 v2 support).

What are the steps (including initial setup) to build EPub-InkPlate?

Based on some poking around my guess is something like the following, but something still isn't right (see below for details):

```
git clone -b 6plusv2-integration https://github.com/turgu1/EPub-InkPlate.git
git clone -b 6plusv2-integration https://github.com/turgu1/ESP-IDF-InkPlate.git
git clone -b v5.3.1 --recurse-submodules https://github.com/espressif/esp-idf.git esp/esp-idf

( cd esp/esp-idf && ./install.sh esp32 )

sed -i -e "s#/home/turgu1/Dev/#${HOME}/#" EPub-InkPlate/CMakeLists.txt

( . ~/esp/esp-idf/export.sh && cd EPub-InkPlate && ./bld_release.sh "$(git describe --tags |cut -c 2- )" 6plusv2 0 )
```

The builds fails with the following error:
```
/home/sascha/EPub-InkPlate/src/models/epub.cpp: In member function 'int16_t EPub::get_item_count()':
/home/sascha/EPub-InkPlate/src/models/epub.cpp:1060:32: error: no matching function for call to 'distance(pugi::xml_named_node_iterator, pugi::xml_named_node_iterator)'
 1060 |   int16_t count = std::distance(it.begin(), it.end());
      |                   ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/string:47,
                 from /home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/locale_classes.h:40,
                 from /home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/ios_base.h:41,
                 from /home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/ios:44,
                 from /home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/ostream:40,
                 from /home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/iostream:41,
                 from /home/sascha/EPub-InkPlate/components/global/src/global.hpp:9,
                 from /home/sascha/EPub-InkPlate/src/models/epub.hpp:6,
                 from /home/sascha/EPub-InkPlate/src/models/epub.cpp:6:
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h:148:5: note: candidate: 'template<class _InputIterator> constexpr typename std::iterator_traits< <template-parameter-1-1> >::difference_type std::distance(_InputIterator, _InputIterator)'
  148 |     distance(_InputIterator __first, _InputIterator __last)
      |     ^~~~~~~~
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h:148:5: note:   template argument deduction/substitution failed:
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h: In substitution of 'template<class _InputIterator> constexpr typename std::iterator_traits< <template-parameter-1-1> >::difference_type std::distance(_InputIterator, _InputIterator) [with _InputIterator = pugi::xml_named_node_iterator]':
/home/sascha/EPub-InkPlate/src/models/epub.cpp:1060:32:   required from here
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h:148:5: error: no type named 'difference_type' in 'struct std::iterator_traits<pugi::xml_named_node_iterator>'
/home/sascha/EPub-InkPlate/src/models/epub.cpp:1064:26: error: no matching function for call to 'distance(pugi::xml_named_node_iterator, pugi::xml_named_node_iterator)'
 1064 |     count = std::distance(it.begin(), it.end());
      |             ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h:148:5: note: candidate: 'template<class _InputIterator> constexpr typename std::iterator_traits< <template-parameter-1-1> >::difference_type std::distance(_InputIterator, _InputIterator)'
  148 |     distance(_InputIterator __first, _InputIterator __last)
      |     ^~~~~~~~
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h:148:5: note:   template argument deduction/substitution failed:
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h: In substitution of 'template<class _InputIterator> constexpr typename std::iterator_traits< <template-parameter-1-1> >::difference_type std::distance(_InputIterator, _InputIterator) [with _InputIterator = pugi::xml_named_node_iterator]':
/home/sascha/EPub-InkPlate/src/models/epub.cpp:1064:26:   required from here
/home/sascha/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/xtensa-esp-elf/include/c++/13.2.0/bits/stl_iterator_base_funcs.h:148:5: error: no type named 'difference_type' in 'struct std::iterator_traits<pugi::xml_named_node_iterator>'
```

Do I need a different version of `esp-idf`?

gigawatts, 2025-06-25:
I second the request for build instructions. I've got a few tweaks I would like to make, as discussed in #18

swlxe, 2025-09-01:
I third, would be helpful!

gigawatts, 2026-01-18:
@turgu1 can we get some basic instructions for how to compile?

gigawatts, 2026-01-21:
I successfully built the firmware, but after flashing it to my InkPlate 6 Plus v2, it seems to fail to init the SPI bus for the SD card, then restarts (over and over). 

Here is the log from serial output:

```
I (604) quad_psram: This chip is ESP32-D0WD
I (605) esp_psram: Found 8MB PSRAM device
I (605) esp_psram: Speed: 40MHz
I (605) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (610) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (618) cpu_start: Multicore app
I (1502) esp_psram: SPI SRAM memory test OK
I (1511) cpu_start: GPIO 3 and 1 are used as console UART I/O pins
I (1511) cpu_start: Pro cpu start user code
I (1511) cpu_start: cpu freq: 240000000 Hz
I (1513) app_init: Application information:
I (1517) app_init: Project name:     EPub-InkPlate
I (1522) app_init: App version:      2.1.0-BETA-I6PV2-20260121-0334Z
I (1528) app_init: Compile time:     Jan 20 2026 21:35:05
I (1533) app_init: ELF file SHA256:  36175a35d...
I (1537) app_init: ESP-IDF:          v5.5.2
I (1541) efuse_init: Min chip rev:     v0.0
I (1545) efuse_init: Max chip rev:     v3.99 
I (1549) efuse_init: Chip rev:         v3.0
I (1554) heap_init: Initializing. RAM available for dynamic allocation:
I (1560) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1565) heap_init: At 3FFC1D90 len 0001E270 (120 KiB): DRAM
I (1570) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1575) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1581) heap_init: At 4009F508 len 00000AF8 (2 KiB): IRAM
I (1587) esp_psram: Adding pool of 4080K of PSRAM memory to heap allocator
I (1594) spi_flash: detected chip: generic
I (1597) spi_flash: flash io: dio
I (1606) main_task: Started on CPU0
I (1616) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1616) main_task: Calling app_main()
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 300, 4MB external flash
Minimum free heap size: 4378236 bytes
Heap summary for capabilities 0x00000C06:
  Totals:
    free 0 allocated 0 min_free 0 largest_free_block 0
I (1636) main: EPub-Inkplate Startup.
W (1666) i2c.master: Please check pull-up resistances whether be connected properly. Otherwise unexpected behavior would happen. For more detailed information, pls
I (1676) PCAL6416: PCAL at address 0x20 has been detected
I (1676) main_task: Returned from app_main()
I (1686) PCAL6416: PCAL at address 0x21 has been detected
E (1776) gpio: gpio_pullup_en(78): GPIO number error (input-only pad has no internal PU)
E (1776) gpio: gpio_pulldown_en(116): GPIO number error (input-only pad has no internal PD)
I (2056) TouchScreen: Software reset OK
I (2056) TouchScreen: Resolution: [1436, 1920]
I (2066) SDCard: Setup SD card
E (2166) spi: spi_bus_initialize(823): no support changing io default level 
E (2166) SDCard: Failed to initialize SPI bus for SD Card.
E (2166) InkPlatePlatform: SDCard setup not completed!
E (2166) main: Inkplate Platform Setup Error. Restarting!
```

gigawatts, 2026-01-21:
Built and flashed from the branch requested. NTP sync worked, and I was able to upload a new epub over Wifi without issues. Touch screen works fine.

gigawatts, 2026-01-22:
Nothing else major at the moment, I'll keep playing with it and let you know!

3 bit mode does cause a full refresh on every update (page turn, menu press, etc), but I think thats listed in your todo.txt, so I've been using 1-bit mode to reduce the refresh time.

I'll continue working on my "Power off / Idle wallpaper" idea, now that I can customize and compile 👍

Thank you SO much for your continued support on this!

closed 2026-02-28 by turgu1